utiasSTARS/pykitti #5: Missing Velodyne calibration data: "KeyError: 'Tr'" (and the solution)

AndreiBarsan, 2017-05-07:
This is just a minor issue which confused me a little at first, and could be nice to have in e.g., the README.
Namely, if one simply downloads the KITTI odometry dataset (e.g., just http://www.cvlibs.net/download.php?file=data_odometry_gray.zip), then the sequence folders will already have calibration files in them, but these files will not have the velodyne-to-rectified-camera-coords information.

One also needs to download the separate calibration data (http://www.cvlibs.net/download.php?file=data_odometry_calib.zip) archive, and overwrite all old `calib.txt` files (which only contain P0-P3 matrices) with the new ones which also contain the `Tr` key, providing the velodyne-to-camera transform matrices.

I was getting a few `KeyError: 'Tr'` before I figured this out. Again, this isn't a major issue, but I just thought I'd point it out in case anyone else was confused.

leeclemnet, 2017-05-12:
Good catch. I updated the readme to point out that the calibration files need to be downloaded separately. (https://github.com/utiasSTARS/pykitti/commit/3f3a7e7d4621546f124bdf5ae3ee012c79f7ffaa)

closed 2017-05-12 by leeclemnet

sunmk2006, 2017-05-31:
hi,i want to use the KITTI odometry dataset to do laser odometry tests , the algorithm needs to use IMU information, but the IMU information is not found in this dataset,Can you help me?