utiasSTARS/pykitti #7: next(dataset.cam0) doesn't work

jinchenglee, 2017-06-27:
It always returns the first frame data no matter how many times I've called it. 

Which version of python does it require? Python 2 or 3? I use Python3.

leeclemnet, 2017-06-27:
`dataset.cam0` is a property that creates a new generator each time you access it, so calling `next(dataset.cam0)` will only ever return the first frame. This is just syntactic sugar to avoid explicitly rewinding the dataset every time you want to iterate through it. The usual thing you'll need to do is something like this:
```python
for image in dataset.cam0:
     do_something(image)
```
which works as you'd expect.

If you need to access arbitrary frames, you have a few options:
1. Pass a `frames` argument to the constructor to load only specific frames like I do [here](https://github.com/utiasSTARS/pykitti/blob/master/demos/demo_raw.py#L23). You can pass an arbitrary list instead of a range if need be.
2. Use the `itertools.islice` method like I do [here](https://github.com/utiasSTARS/pykitti/blob/master/demos/demo_raw.py#L34).
3. Create a list from the generator (this can use up a lot of memory): 
```python
cam0_frames = list(dataset.cam0)
image01 = cam0_frames[1]
```
4. Create another variable to point to a specific instance of the generator, and use `next` like you were trying to do: 
```python
cam0_generator = dataset.cam0
image00 = next(cam0_generator)
image01 = next(cam0_generator)
```

closed 2017-06-27 by leeclemnet

jinchenglee, 2017-06-27:
>    1. Create another variable to point to a specific instance of the

Thanks for the thorough explanation.

It would be helpful you can add these to README.md so others won't ask
again.

leeclemnet, 2017-06-27:
Good idea. I updated the example in the README.md to hopefully make the usage more clear. See: https://github.com/utiasSTARS/pykitti/commit/87035d0c3bef3bd8ce3535a7de63506f8a6ed4a8