utiasSTARS/pykitti #8: demo_odometry.py  KeyError: 'Tr' error

acseckin, 2017-07-01:
after running demo_odometry.py code this error 

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/spyder/utils/site/sitecustomize.py", line 866, in runfile
    execfile(filename, namespace)
  File "/usr/local/lib/python2.7/dist-packages/spyder/utils/site/sitecustomize.py", line 94, in execfile
    builtins.execfile(filename, *where)
  File "/home/acs/Documents/python_projects/DOKTORA/Monocular Odometry/demos/demo_odometry.py", line 23, in <module>
    dataset = pykitti.odometry(basedir, sequence, frames=range(0, 20, 5))
  File "build/bdist.linux-x86_64/egg/pykitti/odometry.py", line 31, in __init__
  File "build/bdist.linux-x86_64/egg/pykitti/odometry.py", line 156, in _load_calib
KeyError: 'Tr'
```

leeclemnet, 2017-07-04:
I'm unable to reproduce this error. 'Tr' appears only in the calib.txt file associated with each sequence. Which sequence are you working with, is calib.txt in the default location, and does it contain a 'Tr' entry?

leeclemnet, 2017-07-28:
Closing due to inactivity.

closed 2017-07-28 by leeclemnet

versatran01, 2018-12-19:
For some reason the version that I downloaded does not have Tr in calib.txt in all sequences.

mehranand1, 2020-03-03:
Error after running demo_odometry.py

In google **Colab, Python3, TPU** , data_odometry_gray.zip 

       KeyError                                  Traceback (most recent call last)
     <ipython-input-16-16db14e38499> in <module>()
     18 # Load the data. Optionally, specify the frame range to load.
     19 # dataset = pykitti.odometry(basedir, sequence)
     ---> 20 dataset = pykitti.odometry(basedir, sequence, frames=range(0, 20, 5))
     21 
     22 # dataset.calib:      Calibration data are accessible as a named tuple

     /usr/local/lib/python3.6/dist-packages/pykitti/odometry.py in __init__(self, base_path, sequence, 
    **kwargs)
     31 
     32         # Pre-load data that isn't returned as a generator
     ---> 33         self._load_calib()
     34         self._load_timestamps()
     35         self._load_poses()

     /usr/local/lib/python3.6/dist-packages/pykitti/odometry.py in _load_calib(self)
    167 
    168         # Compute the velodyne to rectified camera coordinate transforms
    --> 169         data['T_cam0_velo'] = np.reshape(filedata['Tr'], (3, 4))
    170         data['T_cam0_velo'] = np.vstack([data['T_cam0_velo'], [0, 0, 0, 1]])
    171         data['T_cam1_velo'] = T1.dot(data['T_cam0_velo'])

    KeyError: 'Tr' 

@leeclemnet 

[calib.txt(04 sequence)](https://github.com/utiasSTARS/pykitti/files/4278858/calib.txt)

mehranand1, 2020-03-03:
Solved with this file: 
http://www.cvlibs.net/download.php?file=data_odometry_calib.zip