utiasSTARS/pykitti #12: demo_odometry.py "TypeError: list object is not an iterator"

fangchangma, 2017-08-26:
After modifying the `basedir` variable in `demo_odometry.py`, running the script gives me an error
```
Traceback (most recent call last):
  File "demo_odometry.py", line 34, in <module>
    first_gray = next(dataset.gray)
TypeError: list object is not an iterator
```
Platform: Ubuntu 16.04, Python 2.7

leeclemnet, 2017-08-28:
Interesting, I don't have any problem running this code without the `iter()` wrapper.

Two questions:
1. What version of python are you using? 
2. What does `type(dataset.gray)` give you?

fangchangma, 2017-08-29:
1. Python 2.7.12, on Ubuntu 16.04
2. `<type 'list'>`

leeclemnet, 2017-08-29:
This seems to be a Python 2 vs 3 thing. In Python 3 the generators yield a zip object, which is already iterable, so you don't need the `iter` wrapper to call `next` on it. Adding `iter` to the `next` calls in the demo code doesn't seem to break anything, and the list in Python 2 should still be iterable in a for loop, so no problems for normal usage. I'll merge https://github.com/utiasSTARS/pykitti/pull/13 and update `demo_raw.py` to match. Thanks.

closed 2017-08-29 by leeclemnet