utiasSTARS/pykitti #14: Add fileformat kwarg to pykitti.raw()

Mokkaa, 2017-10-06:
Hi,
I'suggest adding an option for fileformat in the call to pykitti.raw(), such that .jpg can be loaded as well.

I've hacked it by adding 
    self.fileformat = kwargs.get('fileformat', 'png')
to the __init__ and replacing '*.png' with
    '*.{}'.format(self.fileformat)
in the generators.

This way other fileformats such as 'jpg' should work as well.

Cheers,
Moritz

leeclemnet, 2017-10-06:
Can you give me an example of why you would want to load JPG files in pykitti.raw? The raw KITTI images are all encoded as PNG.

Mokkaa, 2017-10-17:
True, in some cases it might be beneficial to save memory though by converting the raw data to .jpg

leeclemnet, 2017-10-18:
Alright, it's a minor enough modification that I don't mind making it. You can now pass imtype='jpg' to the raw and odometry constructors. See https://github.com/utiasSTARS/pykitti/commit/838221796cde5d8218f4d6222991abf9936db3b6.

I've updated the package on PyPI as well.

closed 2017-10-18 by leeclemnet