utiasSTARS/pykitti #16: How to convert image coordinates to world coordinates?

jaskaran1, 2017-10-30:
Hello,

I was using pykitti and wanted to know given an image let say from camera 2 for a particular frame and the imu coordinates

How do I convert the pixel coordinates of that image into the world frame?

I was using this line of code for projecting from 2d image coordinates to world coordinates. 
But looks like this doesn't work.

`proj_matrix_cam2_image = data.calib.P_rect_20.dot(data.calib.T_cam2_imu)
proj_matrix_cam2_image_inverse = np.linalg.pinv(proj_matrix_cam2_image)
x,y,z,w=proj_matrix_cam2_image_inverse.dot(np.array([[0,0,1]]).T) #from 2d image homogeneous coordinates to world coordinates
`
Can you tell me where am I going wrong?

This is my first time with the dataset.

PS: My images are from the rectified dataset i.e. image size is (W,H) = (1242,375).

leeclemnet, 2017-10-30:
The IMU-to-world transformations are provided through the oxts property as T_w_imu. If you have points in the IMU coordinate system, you just have to pre-multiply them by T_w_imu to transform them into the world coordinate system. Check the readme for an example of how to access the IMU-to-world transformations.

jaskaran1, 2017-10-30:
I realized later that w in T_w_imu world frame.
Thanks for the clarification.

closed 2017-10-30 by jaskaran1