utiasSTARS/pykitti #18: Erroneous trajectory of egocentric vehicle

jaskaran1, 2017-11-12:
Using the code in the Readme for converting imu to world coordinates, I plot the trajectory of the egocentric vehicle in 2d plane(xy) and the trajectory doesn't seem to be correct in some of the videos.

For instance, in City category for drive 93, I get this plot which doesn't look right at all
[image]

after plotting `point_w` array below

```
point_imu = np.array([0,0,0,1])
point_w = [o.T_w_imu.dot(point_imu) for o in data.oxts]
```

Whereas the same code for drive 15 in Road category gives the correct trajectory of the egocentric vehicle.

MouvementMondial, 2017-11-13:
I think there is a problem with the data. Take a look at this files:

.../2011_09_26/2011_09_26_drive_0093_sync/oxts_data

[readme](https://github.com/yanii/kitti-pcl/blob/master/KITTI_README.TXT)

jaskaran1, 2017-11-13:
The altitude value is negative.

leeclemnet, 2017-11-14:
It looks like the GPS/INS system didn't have a good lock on that particular sequence. Looking at the first OXTS packet in the dataset, we have
```
pos_accuracy=43.329367454418, 
vel_accuracy=0.34225136961012, 
navstat=4, 
numsats=2, 
posmode=0, 
velmode=0, 
orimode=0
```
And this seems to be representative of the entire sequence. So if the position accuracy (I assume this is a standard deviation or similar) is 20-40 meters everywhere, that would explain why your first plot doesn't make sense.

In contrast, looking at `2011_09_30_drive_0018_sync`, the position accuracy is on the order of centimeters, there are significantly more satellites being used, and the pos/vel/orimode flags are different:
```
pos_accuracy=0.067801179930736, 
vel_accuracy=0.017029386365926, 
navstat=4, 
numsats=9, 
posmode=5, 
velmode=5, 
orimode=6
```

jaskaran1, 2017-11-14:
@leeclemnet 
I am trying to create a dataset for trajectory from all the videos in the kitti dataset. What all do I check for a particular sequence(drive) before including its trajectory information?

Do I just check the values for pos_accuracy, vel_accuracy?

leeclemnet, 2017-11-14:
As far as I know, there isn't a list showing which sequences have good GPS and which ones don't. Excluding measurements or entire sequences with a `pos_accuracy` above some threshold seems like a reasonable thing to try.

I suspect `posmode=0` is also a sign that something has gone wrong, but I'm not familiar with what these mode flags actually mean. You could try searching for some documentation about the OXTS packet format.

closed 2017-12-14 by leeclemnet