utiasSTARS/pykitti #21: How to convert velodyne data to rosbag?

AbnerCSZ, 2017-12-17:
How to convert velodyne data( .bin ) to rosbag? Only need to convert velodyne data, without image data.

leeclemnet, 2017-12-17:
There are a couple of rosbag tools listed on the KITTI website: http://www.cvlibs.net/datasets/kitti/raw_data.php

closed 2017-12-17 by leeclemnet

AbnerCSZ, 2017-12-19:
Thanks for your help. I knew some tools listed on this website, such as kitti2bag. And I can use it to convert raw_data to rosbag.
But I can not use it to convert only odometry data in this website:[http://www.cvlibs.net/datasets/kitti/eval_odometry.php](http://www.cvlibs.net/datasets/kitti/eval_odometry.php)
In other words,  this tools only can convert data that meets their formatting requirements( format in [raw_data](http://www.cvlibs.net/datasets/kitti/raw_data.php) .
How can I convert odometry sequences in this website to rosbag?

leeclemnet, 2017-12-19:
The odometry benchmark sequences are just subsets of the raw data. The mapping from odometry sequences to raw sequences can be found in the readme.txt file included with the odometry development kit.

Beyond that, I would recommend opening an issue in the [kitti2bag repo](https://github.com/tomas789/kitti2bag).