utiasSTARS/pykitti #26: Intrinsic matrix

moshanATucsd, 2018-04-01:
Hi, thanks for the code! When we use the matlab devkit to read the kitti camera calibration file, we could use (x - cx)/fx to transform x from image coordinates to a 3D point in camera frame. When we use pykitti, since the row, column indices start at 0 in python instead of 1 as in matlab, should we use (x - cx +1)/fx?

leeclemnet, 2018-04-15:
The definition of camera coordinates is independent of whether the you're working in a zero-indexed or one-indexed language. The pixel at `im(1,1)` in MATLAB and `im[0,0]` in python both contain coordinate (0,0) by definition, usually at the center of the pixel.

moshanATucsd, 2018-04-16:
Hi Lee, 

Thanks for your explanation. Let me make an example to be more clear: say there is a feature, in matlab it is at a = (1, 1), and it will be at a' = (0, 0) in python. Since the calibration file in kitti is the same, say the center is at c = (10, 10), then a - c and a' - c would have different values. Should we use a' - c + 1 instead?

leeclemnet, 2018-04-16:
As far as I know, the KITTI calibration files assume that the top-left pixel is at coordinate (0,0), which is the standard definition for image coordinates. If you're trying to access the pixel value at image coordinate (u,v) in MATLAB, then you will need to use (u+1, v+1) since MATLAB arrays are 1-indexed (i.e., image coordinate (0,0) corresponds to array indices (1,1) ). This has no bearing on how the image coordinates are defined for the purposes of calibrating the camera model. It only impacts how you access the image data.

moshanATucsd, 2018-04-17:
Hi Lee,

I see, thanks for your clarification!

closed 2018-04-17 by moshanATucsd