utiasSTARS/pykitti #29: Can't use the "dataset.get_velo(idx)" method from pykitti

liemwellys, 2018-07-17:
Hi, How do you set up the folder for dataset downloaded from KITTI website? Because when I try to load the velodyne point using method dataset.get_velo(0), it has error "Index out of range".

my configuration is like this:
basedir = 'data'
date = '2011_09_26'
drive = '0005'

leeclemnet, 2018-08-01:
The [KITTI dataset IJRR paper](http://www.cvlibs.net/publications/Geiger2013IJRR.pdf) has details on the file structure. When you unzip the data files, it should already be set up properly.

Have you tried changing your basedir to an absolute path?

canorbal, 2018-10-08:
@liemwellys could you please explain what you mean by set up properly? I've downloaded data_odometry_color.zip, data_odometry_poses.zip, data_odometry_calib.zip and devkit_odometry.zip. After that I've unzipped everything but got the same error

thx for any help!

leeclemnet, 2018-10-17:
@canorbal When you unzip the files they should all merge into a single directory structure laid out as specified in the paper linked in https://github.com/utiasSTARS/pykitti/issues/29#issuecomment-409591661. As long as your basedir is pointed to this folder, you shouldn't encounter any problems. You may have to use an absolute path for the basedir.

closed 2019-03-01 by leeclemnet