utiasSTARS/pykitti #32: Evaluate BEV detections

Tgaaly, 2018-07-30:
I'm using the KITTI cpp code (evaluate_object_3d_offline.cpp) to evaluate my birds eye view bounding box detections. The code filters the detections into easy, moderate and hard by the height of the detected object's bounding box in the image space (i.e. pixels). I do not have image detections at all and so all my detections are being ignored (lines 447-452). I'm experimenting solely on BEV data. Any idea how to evaluate ONLY BEV detections (without any corresponding image detections). Is there a reference somewhere for this?

leeclemnet, 2018-08-01:
This is not the right place for this discussion. I suggest contacting the maintainers of the KITTI dataset for questions like these: http://www.cvlibs.net/aboutme.html

closed 2018-08-01 by leeclemnet

Petros626, 2025-05-03:
@Tgaaly hey can you help me?
I'm trying to get the same results with the predictions of my model, but struggle for now.