utiasSTARS/pykitti #34: Creation of T_camX_velo all use R_rect_00

keenan-burnett, 2018-10-16:
This is what is instructed in the paper, but this seems like a typo?
I'm projecting the pointcloud onto the image plane for T_cam2_velo and I'm get slightly better results when using R_rect_20.

leeclemnet, 2018-10-17:
I always thought this was strange, but the MATLAB devkit also uses `R_rect_00` for all the velodyne-to-camera transformations. What do you mean when you say you get better results? Do you also get improvements for cam1 and cam3?

keenan-burnett, 2018-10-17:
Which MATLAB devkit are you referring to? It seems that the velodyne points, when projected onto camera 2, align slightly better when R_rect_20 is used. I looked at the other cameras as well. Similarly for camera 3, the point projection works better when using R_rect_30. Surprisingly, using P_rect_30, with T_cam2_velo and R_rect_20 resulted in the best projection for camera 3.

projection_onto_cam2_with_r_rect_00 (P_rect_20)
[image]

projection_onto_cam2_with_r_rect_20 (P_rect_20)
[image]

projection_onto_cam3_with_t3_r_rect_00 (P_rect_30)
[image]

projection_onto_cam3_with_t3_r_rect_30 (P_rect_30)
[image]

projection_onto_cam3_with_t2_r_rect_20 (P_rect_30)
[image]

leeclemnet, 2018-10-17:
I'm referring to the raw data devkit provided on the [KITTI website](http://www.cvlibs.net/datasets/kitti/raw_data.php). If you look at the file `loadCalibration.m`, they've used `R_rect_00` everywhere. 

Looking at the [KITTI dataset IJRR paper](http://www.cvlibs.net/publications/Geiger2013IJRR.pdf) again, this seems to be because the way they build up the velodyne to camera transformation is with the sequence velodyne-->unrectified cam0 coordinates-->rectified cam0 coordinates-->rectified camN coordinates. The transformation from unrectified cam0 coordinates-->rectified cam0 coordinates should always use `R_rect_00`. 

I think the improvements you're seeing might be coincidental, or else an artifact of imperfect calibration.

keenan-burnett, 2018-10-17:
Okay, sounds good. Feel free to close.

closed 2018-10-17 by leeclemnet