utiasSTARS/pykitti #36: How to read the tracklets and get the box on image? (for both 2d and 3d)

jinfagang, 2018-10-30:
As title suggested

MOHAMEDELDAKDOUKY, 2019-02-27:
Same problem here!

leeclemnet, 2019-03-01:
Currently we only support the raw and odometry datasets. Support for the tracking dataset would be a nice addition if someone in the community wanted to tackle it and submit a PR.

MOHAMEDELDAKDOUKY, 2019-03-01:
I've managed to do it and will make it available soon.  thanks :)

leeclemnet, 2019-03-01:
Apparently I forgot about https://github.com/utiasSTARS/pykitti/pull/28, which added support for the tracking dataset. I haven't used it personally. Does it do what you need, or have you added something on top of it to get the bounding boxes?

MOHAMEDELDAKDOUKY, 2019-03-01:
In fact, I did not use it. Instead, I build some wrapping on top of Christian Herdtweck XML parser.
I'm trying to add a Dynamic/Static annotation to the tracklets.

LukeAI, 2019-06-06:
Hey guys,
I want to be able to read the raw tracklets and extract the projected 2d bounding boxes and corresponding rgb image. I understand that this is possible with these utils but How can I do this?
@MOHAMEDELDAKDOUKY @leeclemnet

closed 2019-08-28 by jinfagang