utiasSTARS/pykitti #42: Projection from pointcloud on image plane

langelesl, 2019-03-01:
Hello. I tried to use [x, y, z, 1] = T_cam2_velo * [x_velo, y_velo, z_velo, 1] and [u,v,1] = K_cam2 * [x/z, y/z, 1] to calculate the coordinates of point cloud on the image plane of camera 2 using .bin file of point cloud in raw data . The range of values for  u and v is extremely large in the result I got (10^-8 < u < 10^8 , 10^-7 < v < 10^7 ). That looks really weird. Is that a normal range of u, v-value for the point cloud in KITTI? Or maybe I used the wrong equations here?
Thanks for your help.

leeclemnet, 2019-03-06:
Are you using * or .dot() here? It should be .dot() to do the matrix multiplication correctly. Otherwise I would suggest inspecting T_cam2_velo, K_cam2 and the x,y,z coordinates you're using to make sure they have sensible values.

langelesl, 2019-03-11:
I used .dot(). Since the equations are suitable here, maybe there is something wrong with the coordinates values I used. I will check it later. Thank you for your replay.

closed 2019-03-11 by langelesl