utiasSTARS/pykitti #48: A question about Velodyne Data Label

iris0329, 2019-06-11:
Hi,

Thanks for sharing such good work !
I found in [intel-isl/Open3D-PointNet2-Semantic3D](https://github.com/intel-isl/Open3D-PointNet2-Semantic3D) project, the author use your work to handle KITTI Velodyne data.
It's a segmentation work, each point has a label to show which class it belongs. But I found velodyne point data only has 4 channels (x, y, z, reflection). So do you have access to the ground truth label of each point? 

Thanks for your help !

slinshady89, 2019-07-08:
Probably this will help you
http://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d

xiongyiheng, 2021-05-03:
> Hi, @leeclemnet
> 
> Thanks for sharing such good work !
> I found in [intel-isl/Open3D-PointNet2-Semantic3D](https://github.com/intel-isl/Open3D-PointNet2-Semantic3D) project, the author use your work to handle KITTI Velodyne data.
> It's a segmentation work, each point has a label to show which class it belongs. But I found velodyne point data only has 4 channels (x, y, z, reflection). So do you have access to the ground truth label of each point?
> 
> Thanks for your help !

@iris0329 Hi, have you found out how to solve the issue?