utiasSTARS/pykitti #50: point projection gives different results comparing to kitti depth dataset

liruikangyk, 2019-07-24:
In my project I have to downsample the KITTI velodyne lidar data to 16 lines (KITTI depth dataset provides original 64-line projected png files of cam02 and cam03)
so I need to 
(1) downsample the 64-line bin lidar data to 16-line (this has already been done)
(2) use the calib parameters to project the point from bin files to 2d pictures of cam02 and cam03
however, I met a problem in step 2: I find that points in my projected pngs look a little bit shifted
so I tried to test my projection procedure : project the original 64-line lidar data to cam02 png, and merge my projected image with the KITTI-provided projected image of cam02.
I can see that most of the points are at the same pixel location, but the points in the very right and very left of the image are shifted from the correct location (especially points of the objects that are near from the camera, ) 

[image]

my code is here:
[image]

trns1997, 2020-02-22:
Hi @liruikangyk may I ask how did you go about downsampling the 64-line bin lidar data from the kitti dataset to 16-line?

hulsmeier, 2020-10-06:
> Hi @liruikangyk may I ask how did you go about downsampling the 64-line bin lidar data from the kitti dataset to 16-line?

did you ever figure out a way?

EthanZhangYi, 2021-03-17:
@liruikangyk 
Hi, 
I also meet the same problem. I tried several coordinate transformation methods but the generated depth map is still different from those in **proj_depth/velodyne_raw** of KITTI Depth Dataset.

Do you figure out the reason for this mismatching?
I am also interested in the method you used to down-sample the 64-beam point cloud. Could share some information on it?

Thanks~

minghaohsu410168, 2021-04-03:
@liruikangyk ,

How do you downsample the lidar data from 64 line to 16 lines? Please give me some advice, thank you.