utiasSTARS/pykitti #62: Is there correspondence between raw and odometry videos?

NagabhushanSN95, 2021-03-19:
Are the 22 sequences available in the odometry dataset, taken from the raw dataset? Any idea?

brucemuller, 2021-05-14:
The 11 training sequences are, as can be seen at the end of the readme for the odo set. The exception is one cannot be found in the raw:

00: 2011_10_03_drive_0027 000000 004540 FOUND
01: 2011_10_03_drive_0042 000000 001100 FOUND
02: 2011_10_03_drive_0034 000000 004660 FOUND
03: 2011_09_26_drive_0067 000000 000800 CANNOT FIND
04: 2011_09_30_drive_0016 000000 000270 FOUND
05: 2011_09_30_drive_0018 000000 002760 FOUND
06: 2011_09_30_drive_0020 000000 001100 FOUND
07: 2011_09_30_drive_0027 000000 001100 FOUND
08: 2011_09_30_drive_0028 001100 005170 FOUND
09: 2011_09_30_drive_0033 000000 001590 FOUND
10: 2011_09_30_drive_0034 000000 001200 FOUND

NagabhushanSN95, 2021-05-14:
Cool, thanks.

closed 2021-05-14 by NagabhushanSN95